damellis/ESP #402: Program crashing after training is successful

dukeboy86, 2017-12-12:
Hello,

I hope this is not a silly question or issue report. I am currently using the version ESP-Gestures-Win-20161028 and everything goes alright at the beginning. I am able to calibrate the sensor successfully and then after inputting 3 or 4 samples of two different movements in slots 1 & 2 I do the training which shows to be successful. The movements that I am training are standing up and sitting down on a chair while the sensor is stuck to the back of my neck. I'm using an Adafruit Feather M0 board.

However, after the training is done and I want to check if the classifier works correctly I go to the "Prediction" tab the program crashes and I have to force close it and restart it again. I don't know if I'm missing something in the steps prior or what could be causing the crash. After this happens I open a serial terminal and my board keeps sending sensor information, so the problem is somewhere in the program. 

Thanks for your help.

sp436857, 2018-01-08:
I am experiencing the same problem. I would like to know where the problem is ? 

Thanks for your help.. !

kartik9k, 2018-02-23:
Experiencing the same problem! Would like the contributors to share a fix soon!

mrrobot2802, 2019-06-18:
did someone find solution to this?

thanks!