damellis/ESP #404: cannot run example code on processing

kamilalfian, 2018-03-10:
So I got a "not receiving data" symptom which is why I decided to use esp mode on processing. While trying the gesture example code I got this error:

C:\Users\LENOVO\Documents\Processing\modes\ESPMode\core>link.exe /ERRORREPORT:PROMPT /OUT:"C:\Users\LENOVO\AppData\Local\Temp\user_accelerometer_gestures2346038532781405631temp\bin\ESP_debug.exe" /INCREMENTAL /NOLOGO  openframeworksLib_debug.lib ESPlib_debug.lib grt.lib "cairo-static.lib" "pixman-1.lib" libpng.lib msimg32.lib OpenGL32.lib GLu32.lib kernel32.lib setupapi.lib Vfw32.lib comctl32.lib glut32.lib rtAudioD.lib videoInputD.lib libfreetype.lib FreeImage.lib qtmlClient.lib dsound.lib user32.lib gdi32.lib winspool.lib comdlg32.lib advapi32.lib shell32.lib ole32.lib oleaut32.lib uuid.lib glew32s.lib fmodex_vc.lib glu32.lib ssleay32MD.lib libeay32MD.lib crypt32.lib PocoFoundationmdd.lib PocoNetmdd.lib PocoUtilmdd.lib PocoXMLmdd.lib Ws2_32.lib tess2.lib glfw3.lib winmm.lib kernel32.lib user32.lib gdi32.lib winspool.lib comdlg32.lib advapi32.lib shell32.lib ole32.lib oleaut32.lib uuid.lib odbc32.lib odbccp32.lib /LIBPATH:library/ /LIBPATH:library/openFrameworks/libs/glut/lib/vs/Win32 /LIBPATH:library/openFrameworks/libs/glfw/lib/vs/Win32 /LIBPATH:library/openFrameworks/libs/rtAudio/lib/vs/Win32 /LIBPATH:library/openFrameworks/libs/FreeImage/lib/vs/Win32 /LIBPATH:library/openFrameworks/libs/freetype/lib/vs/Win32 /LIBPATH:library/openFrameworks/libs/quicktime/lib/vs/Win32 /LIBPATH:library/openFrameworks/libs/fmodex/lib/vs/Win32 /LIBPATH:library/openFrameworks/libs/videoInput/lib/vs/Win32 /LIBPATH:library/openFrameworks/libs/cairo/lib/vs/Win32 /LIBPATH:library/openFrameworks/libs/glew/lib/vs/Win32 /LIBPATH:library/openFrameworks/libs/glu/lib/vs/Win32 /LIBPATH:library/openFrameworks/libs/openssl/lib/vs/Win32 /LIBPATH:library/openFrameworks/libs/Poco/lib/vs/Win32 /LIBPATH:library/openFrameworks/libs/tess2/lib/vs/Win32 /LIBPATH:library/openFrameworks/libs/boost/lib/vs/Win32 /NODEFAULTLIB:PocoFoundationmdd.lib /NODEFAULTLIB:atlthunk.lib /NODEFAULTLIB:msvcrt /NODEFAULTLIB:libcmt /NODEFAULTLIB:LIBC /NODEFAULTLIB:LIBCMTD /MANIFEST /MANIFESTUAC:"level='asInvoker' uiAccess='false'" /manifest:embed /DEBUG /PDB:"C:\Users\LENOVO\AppData\Local\Temp\user_accelerometer_gestures2346038532781405631temp\bin\ESP_debug.pdb" /SUBSYSTEM:CONSOLE /TLBID:1 /DYNAMICBASE:NO /NXCOMPAT /IMPLIB:"C:\Users\LENOVO\AppData\Local\Temp\user_accelerometer_gestures2346038532781405631temp\bin\ESP_debug.lib" /MACHINE:X86 /SAFESEH /ignore:4099 "C:\Users\LENOVO\AppData\Local\Temp\user_accelerometer_gestures2346038532781405631temp\obj\user.obj" 
'link.exe' is not recognized as an internal or external command,
operable program or batch file.
Exited with: 1

PS: I'm a layman so please consider this when replying, thank you.

nebgnahz, 2018-03-30:
The Processing extension may be problematic. Have you tried to build the project on its own?

kamilalfian, 2018-03-31:
Uhh it's alright now i fixed it a long time ago. Apparently I just need to modify my own arduino code by putting some semicolons and tab in the printf part of the code so it can synchronize with the esp code.

closed 2018-03-31 by kamilalfian