damellis/ESP #405: Is this Project dead?

lordgort, 2018-03-22:
Is someone still working with this?

nebgnahz, 2018-03-30:
At this point, we are not actively developing new features. The code still runs (I just checked) but that is with an old version of every dependency (openFrameworks, GRT, ofx plugins). If you have a specific question or issue, I'd be happy to answer/help.

The main maintainers (me and @damellis) currently have other commitments. I might be able to upgrade everything around May; but at this point, I do not have enough bandwidth for this. Apologies.

lordgort, 2018-04-03:
Oh great to see that someone still reads this :-D
I am new to Linux (raspbian, armv6) and it was really hard to compile ESP on my Raspberry Pi Zero W, but i did it. 
I spend 2 weeks, a few hours daily with it @_@. 
But now when starting ESP i just get a black window -.- 
You dont have a precompiled version for an arm system i bet? XD